supermerill/SuperSlicer #3126: Crash when slicing

fungreenfox, 2022-08-25:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 46bit 48gb ram, 3.7ghz quad core.

### Behavior
* _Describe the problem_ Slicing then crash. 
* _Steps needed to reproduce the problem_ Just slice and crash.
* _Expected Results_ NO crash.
* _Actual Results_ Bad.
  * _Screenshots if useful_ 

_Is this a new feature request?_ No, its a request for fix a crash issue in superslicer.
 Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occur
[PictureJPG.zip](https://github.com/supermerill/SuperSlicer/files/9424528/PictureJPG.zip)
[PictureJPG.zip](https://github.com/supermerill/SuperSlicer/files/9424533/PictureJPG.zip)
s
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[bagplade2.zip](https://github.com/supermerill/SuperSlicer/files/9424503/bagplade2.zip)
_

neophyl, 2022-08-25:
So what version of super slicer are you using ?

supermerill, 2022-08-28:
fixed in next nightly.
You can also reset/remove the print custom variable (in print->notes tab)