supermerill/SuperSlicer #3127: Weird illogical printing order of perimeters

kb-elmo, 2022-08-25:
### What happened?

On some of my prints I noticed that the slicer is printing the outside perimeters in a really weird order which creates a ton of completely unnecessary travel moves and causes ugly printing artifacts on the surface of the print that could easily be avoided.

In the example file this behaviour starts on layer 31.
On all previous layers the outside perimeter is printed in a correct logical order.

Screenshots of the sliced part with the erratic behavior:
[image]
The extruder reaches the end inside perimeter
[image]
Instead of continuing with the outside perimeter in the same move it now makes a retraction, moves to a completely different section of the part, then returns to the outside perimeter on the seam position.

This creates a completely unnecesary retraction scar in the outside perimeter.

### Project file & How to reproduce

[top_right.zip](https://github.com/supermerill/SuperSlicer/files/9424652/top_right.zip)

### Version

2.4.58.4

### Operating system

Windows 10

### Printer model

Custom mk3 style with direct drive v6 hotend

kb-elmo, 2022-08-25:
FWIW this does not happen with the new Arachne Beta version.

That one gets the order of the perimeters right.

VanessaE, 2023-08-30:
No movement on this?  Perimeter order and travels are just illogical at present.

VanessaE, 2023-09-01:
Here's another example of this issue:

[image]

The way Superslicer is doing things now, it starts in the middle of the part and works its way leftward, laying perimeters in sets of three for the round holes numbered 1, 2, and 3 in this image, in that order.  Then it travels  across the print to do the the set of three perimeters that make up no. 4, then the two internal perimeters around hole no. 5 (it initially \*skips* the hole's external perimeter).

Then, it jumps back to the left a bit, and starts laying down the external perimeters for the "bowtie"-shaped external perimeter loops, in order 6-7-8 (the internal perimeters that go with them were printed earlier, when it was doing the round holes).  Then, it jumps BACK to the right to do bowtie no. 9, followed by that external loop for hole no. 5 that it skipped earlier (which I've marked as "10" here).

THEN it jumps all the way back to the left and starts the part's outermost perimeter at the point marked "11", proceeding counterclockwise around the part. Lastly, it lays down all those white gap fills in counterclockwise order, starting with the bit closest to where it started the outermost perimeter.

Most of this wouldn't normally be too big of an issue, except that this part has relatively thin walls and lots of holes, and each time the slicer creates a travel, it has a tendency to either travel along outward-facing lines, even ones it has only *just* printed, or it'll travel along spaces that will soon be used for outward-facing lines.  Since these are non-retract travels, that means it'll be leaving ooze along an outward-facing perimeter, which ruins the surface quality.

I would have thought the most logical order would have been to alternate between the round holes' perimeters and the perimeters that make up the bowtie shapes, working from one end of the part to the other, for example in the order 5-9-4-6-1-7-2-8-3 in the image, then printing the outermost perimeter and the gap fills more or less like it does now.

I have tinkered with every setting I can think of that seems like it could influence this, but nothing helps.

Something that may be relevant though -- and this is NOT shown in the image above or present in the file below -- at some point in my tinkering, my filament setting changed unintentionally, which somehow caused the print order to switch to left-to-right instead of the above right-to-left, though the print still started in the middle of the object.  I can only guess that was because the newly-selected filament has a lower maximum overlap than the one I'd been using (65% vs. 90%, if it matters).

Model in question: 
[Technic beam, modern, 5L (with supports).3mf.zip](https://github.com/supermerill/SuperSlicer/files/12491978/Technic.beam.modern.5L.with.supports.3mf.zip)

(for reference, the view shown above is at layer 10, 2.0 mm up on the part).

VanessaE, 2023-11-27:
While it may work "as intended", it should be noted that the Arachne mode has a more logical ordering.

supermerill, 2023-11-27:
> While it may work "as intended", it should be noted that the Arachne mode has a more logical ordering.

I was answering to op.

For your separate issue, yes, it's sub optimal.
Just checked:
- ps 2.5 classic: print external before internal, worst.
- ps 2.5 arachne: optimal
- susi classic: like you said, sub optimal but right order.
- susi arachne: less optimal than ps arachne, but better than susi classic

I'll look again after the merge.