supermerill/SuperSlicer #3128: Crash when sllicing

shortbus, 2022-08-25:
### What happened?

Describe the problem: Slicing then crash.
Steps needed to reproduce the problem: Just slice and crash.
Expected Results: NO crash.
The following has crash logs from both versions 2.4.58.4 and 2.5.59.0. I tried both and both crash. I tried 2.3.57.12 and it slices, but of course I want the new features.
[Mac_Crash_Logs.zip](https://github.com/supermerill/SuperSlicer/files/9425903/Mac_Crash_Logs.zip)
The following is what I was slicing.
[plate4.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9425891/plate4.3mf.zip)

Thanks for all the work you do.

### Project file & How to reproduce

Open the 3mf file and click slice.
It will crash every time.
I have tried slicing over and over and one previous plate it did slice without crashing, but this plate continues to crash.

### Version

2.4.58.4 and 2.5.59.0

### Operating system

MacOS 12.3.1

### Printer model

CR6MAX, running klipper.

kingzmx, 2022-08-27:
The same

shortbus, 2022-08-27:
FYI - I tried version 2.4.58.3 and it does slice the same file. I hope that information helps.

realien-OG, 2022-08-28:
same for me when slicing a cylinder with property overrides for number of layers / infill.

shortbus, 2022-08-31:
I see the bug fix update, THANKS!
Also, if it helps, I noticed in 2.4.58.3 the file would slice if I changed Print Settings > Infill > Infill > Solid from "Rectilinear (filled)" to "Rectilinear".
I tried this same thing with 2.4.58.4 and superslicer still crashed. Looking forward to the next release.

closed 2024-02-11 by supermerill