supermerill/SuperSlicer #3129: Object modifier for perimeter order broken

kb-elmo, 2022-08-25:
### What happened?

Creating a modifier object to change the perimeter order to "external perimeters first" creates new perimeters inside the object

[image]
[image]

The part should look like this instead and the modifier should not add new perimeters.
[image]

### Project file & How to reproduce

[testblock.zip](https://github.com/supermerill/SuperSlicer/files/9425991/testblock.zip)

### Version

2.4.58.4

### Operating system

Windows 10

neophyl, 2022-08-25:
Yes, this is normal and how it works. It splits it into 2 volumes that are then sliced. It’s been reported as a ‘bug’ multiple times here and on the PRUSA forum.

kb-elmo, 2022-08-25:
How is this "intended behavior"?

A modifier should only change the settings of the print. Not split up the object into multiple (useless) pieces.  
This completely defeats the purpose of a modifier and makes it practically useless.

If it doesn't work why does the feature even exist then?  
In this current form it's useless for anything other than modifying infill density.

neophyl, 2022-08-25:
Take it up with PRUSA.
 If you reduce the modifier so it doesn’t poke out the top/bottom then it doesn’t put the extra perimeters visibly and you still get ones inside so it does work. You just have to use them within the limitations they have. 
 Go add your support to the Many existing issues from dissatisfied people who agree with you. Meanwhile until they way they work is completely re-written you need to learn how to use them.

kb-elmo, 2022-08-25:
> If you reduce the modifier so it doesn’t poke out the top/bottom then it doesn’t put the extra perimeters visibly

This doesn't help at all as it still breaks the part, making it unusable.

> You just have to use them within the limitations they have.
> Meanwhile until they way they work is completely re-written you need to learn how to use them.

This isn't a "limitation". It's simpy a broken by design implementation that does not work in the way a user would expect it to.  
And there honestly isn't much to learn about a broken feature imho.

But since this seems to be a Prusa related issue I'll close this here.

closed 2022-08-25 by kb-elmo