supermerill/SuperSlicer #3130: FR: Length Extension for Bridge Flow

CCS86, 2022-08-25:
Because there is always a time delay between a requested increase in flow rate, and the actual increase in flow rate, I think having a configurable extension distance to bridge flow rate could help out.

The current behavior begins the increase flow rate right at the point of the bridge start, so it is probable that this causes a dip in the desired flow rate right at the start of the bridge:

[image]

qwewer0, 2022-08-28:
Like the `bridged_infill_margin`, just for the bridged perimeters?

Anyway, good idea.

CCS86, 2022-08-28:
> Like the `bridged_infill_margin`, just for the bridged perimeters?
> 
> Anyway, good idea.

I guess the result would be similar enough that you could have that same setting govern the 'D' distance I drew in above, to avoid adding a new parameter.

supermerill, 2022-08-29:
a kind of pressure advance?

CCS86, 2022-09-16:
> a kind of pressure advance?

I guess you could think of it like that. Obviously normal pressure advance will still take affect when the extrusion rate jumps. But, since you might have both a jump up in speed, and in extrusion cross section, it can be a giant jump up in flow rate. To start that change after the nozzle has already left the support of the previous layer, seems like it will be more likely to disconnect.

CCS86, 2022-09-16:
This also shows how on an overhang, the bridge flow can start after the end of support:

[image]

CCS86, 2022-09-16:
Or possibly just apply the existing anchor distance setting to the bridge/overhung perimeters, so the increased flow position matches:

[image]

CCS86, 2022-09-16:
I am going to run this test. As generated, vs hand edited to extend the bridged perimeter by 1mm on either side:

[image]

CCS86, 2022-09-16:
Test results (original on left / modified on right):

[image]

qwewer0, 2022-09-17:
@CCS86 Coud you try your test with `Bridge flow baseline` (`bridge_type`) set to `Layer height`?

CCS86, 2022-09-17:
I don't think there would be much of a difference with this modification, because there wouldn't really be a change in flow. 

After a lot of bridge testing, I found the "nozzle diameter" flow calculation to perform much better.