supermerill/SuperSlicer #3134: Vase mode allows more than 1 perimeter

MrBlackmidi, 2022-08-26:
**Bug description**
I'm not sure if this is intended or a bug, but when i activate vase mode, it seems to allow more than 1 perimeter than restricted to only 1

**Expected behavior**
Vase mode does not allow more than 1 perimeter, although i'm not sure if this is intended or not

neophyl, 2022-08-26:
So which version of Super Slicer are you reporting this issue against ?  Which operating system ?

Where is your attached zipped 3mf project file as requested when you open a new issue.  There are like a million settings and its quite possible one or a combination of them is leading to this behavior.

I just tried in 2.5.59 on Win 10 and when selecting spiral vase option it pops up the expected window with a list of settings it needs to change.  While it doesnt change the configured perimeters it does ignore that value and just slice with a single perimeter.

Unless you are using Arachne as the slicing engine ?  That just plain doesn't work with vase mode.  Even the Prusa implementation that SuSi is built on doesn't work with that.

If it is the arachne engine then it would probably be a good idea to set the slicing back to classic automatically when vase is ticked.

MrBlackmidi, 2022-08-27:
ive been using arachne, i didnt know that would happen

contrapuntal, 2022-08-27:
A recent change, [https://github.com/supermerill/SuperSlicer/pull/2998](url), allows multiple perimeters on the bottom layers. Any layers above that are still sliced with single perimeters.

closed 2022-10-23 by MrBlackmidi