supermerill/SuperSlicer #3135: Error: No bitmap handler for type 17 defined

VrtlBrown, 2022-08-26:
### Version
2.4.58.4

### Operating system type + version
macOS 12.5.1

### Behavior
Open SuperSlicer and get an error message:

No bitmap handler for type 17 defined
Details
11:45:23: can't open file '/Applications/SuperSlicer.app/Contents/MacOS/../Resources/splashscreen/SuperSlicer icon' (error 2: No such file or directory)
11:45:23: Failed to load image from file "/Applications/SuperSlicer.app/Contents/MacOS/../Resources/splashscreen/SuperSlicer icon".
11:45:23: no bitmap handler for type 17 defined.

If I click 'OK' I can continue to use the app.

I've tried to install all the extra stuff via HomeBrew to do an update of SuperSlicer and this started appearing after that
[image]

supermerill, 2022-08-28:
A bit strange.
What are the item in this box?
[image]

If you don't have access to that, what have you in your SuperSlicer.ini -> splash_screen_editor line? (in your configuration folder , you can open it from help -> showconf folder)

VrtlBrown, 2022-08-28:
Hi, it’s just on default for me. Here’s the screen shot:

Thanks,

Chris

supermerill, 2022-08-28:
there is still no screenshot. Don't use emails if you want to add one, use the webste.

supermerill, 2022-08-28:
can't reproduce and I can't see how that can happens.
Can you zip & send me your configuration folder (available from help->show configuration folder)?

VrtlBrown, 2022-09-01:
Sorry I didn't see your last couple messages come through. Here are the files:
[image]
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9472517/SuperSlicer.zip)

jimmyD30, 2022-10-16:
> ### Version
> 2.4.58.4
>
> ### Operating system type + version
> macOS 12.5.1
>
> ### Behavior
> Open SuperSlicer and get an error message:
>
> No bitmap handler for type 17 defined Details 11:45:23: can't open [...]

Hi, to stop the error you can rename a file in the package at the location noted in the error message, use a jpeg file and name it 'SuperSlicer icon' with no file extension (i renamed the 'cars.jpg' file located in the 'splashscreen' folder.