supermerill/SuperSlicer #3136: Failed Bridge Detection

CCS86, 2022-08-26:
### What happened?

I can't seem to do anything to get this bridged area properly detected:

[image]

### Project file & How to reproduce

[Bridgefail.3mf.txt](https://github.com/supermerill/SuperSlicer/files/9433025/Bridgefail.3mf.txt)

### Version

2.5.59.0

### Operating system

Windows 10 x64

### Printer model

UMO+

Creo2005, 2022-08-26:
Incorrect slicer settings
[image]

neophyl, 2022-08-26:
[Bridgefail_neo.zip](https://github.com/supermerill/SuperSlicer/files/9433812/Bridgefail_neo.zip)

Yep.  Firstly you are removing all perimeters from bridges.  But thats sort of ok.  biggest issue is you have bottom layers set to zero.  A bridge is a type of bottom layer.
[image]

Try the attached zipped project.

CCS86, 2022-08-26:
Man, I am 0 for 2 today!

I normally run at least 1 bottom layer for this purpose, but must have changed the setting for another print and missed it. 

Thank you!

closed 2022-08-26 by CCS86