supermerill/SuperSlicer #3137: Skirt Missing in Gcode Preview

CCS86, 2022-08-26:
### What happened?

Skirt shows up in sliced preview:

[image]

But disappears from the gcode preview, despite being in the gcode:

[image]

### Project file & How to reproduce

[Skirt-missing.3mf.txt](https://github.com/supermerill/SuperSlicer/files/9433487/Skirt-missing.3mf.txt)

### Version

2.5.59.0

### Operating system

Windows 10 x64

### Printer model

UMO+

neophyl, 2022-08-26:
Well its there when i use your project file in the same version and same OS.

Basic question but you do have skirt enabled on the key dont you ?  The images you posted are cropped to cut out the key so I cant tell.

CCS86, 2022-08-26:
Ooof! 

I didn't even realize that you could click features in the key to show/hide. Must have been an errant mouse click, thanks!

closed 2022-08-26 by CCS86