supermerill/SuperSlicer #3138: Add First and Top Fill Flow Settings to the Filament Overrides List on the Filament Tab

courtross2010, 2022-08-26:
When working with different hotends and nozzles I always end up creating custom print profiles for the filament that I'm using.  The reason for this is that I end up needing to fine-tune the First/Top layer flow rates to get the print dialed in.  Because of this, I end up having to have way more print profiles than I need.  

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The solution I would propose to solve this issue is to add the "Top Fill" and "First Layer" flow fields to the "Filament Overrides" list on the Filament tab of Superslicer.  This way you can enable the override and set a value that works for the specific filament.

Doing this would allow a user to have One Profile that would work for multiple hotends, and multiple nozzles, without having a custom print profile for each filament.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The alternative to this is to create custom print profiles for each filament.  An example would be that I have an E3D Revo using a 0.4 nozzle.
- Printer Profile: Voron (Revo 0.4)
- Filament Profile: KVP ABS
- Print Profile: Standard (0.2 Layer Height) with "Top Fill" set to 98 and "First Layer" set to 100

The above print profile works great for the 0.4 nozzles.  The issue comes up when I switch the Revo to a different nozzle.
- Printer Profile: Voron (Revo 0.6)
- Filament Profile: KVP ABS
- Print Profile: Standard (0.2 Layer Height) with "Top Fill" set to 95 and "First Layer" set to 96

When I print a model using the same filament/print profile as the 0.4 nozzles I run into issues with the first/top layers.  In this scenario, I end up getting over extrusion on the first and top layers.  So, I can either update the print profile and then print.  Then I have to roll back the setting if I switch nozzles.  That gets confusing and I always end up saving changes that mess with the next print.

So right now I have a "Standard (0.2 Layer Height" profile for each nozzle that I have.  This ends up being 4 "Standard" profiles.  Then I have the Draft, Detail, and High detail print profiles that I use.  That's a whole lot of print profiles that the only difference between them is the flow setting for the first and top layers.

By adding the two settings to the Filament Overrides list I can get rid of all the extra profiles.  Then have the Standard, Draft, Detail, and High Detail profiles that work with any nozzle.  I can then just create a filament profile for the nozzle.  In the end a lot less confusion.

Hopefully, this all makes sense.  Maybe I'm the only one who thinks this would be amazing but I'm sure other people would use it.

supermerill, 2022-08-31:
#171

supermerill, 2022-08-31:
note that you can create a custom gcode macro to simulate the behavior.
in your filament custom variable:
```
fil_top_flow=110
fil_first_layer_flow=110
```
in "Start gcode":
```
{default_int(fil_top_flow,100})
{default_int(fil_first_layer_flow,100})
{if exists(fil_first_layer_flow)}M220 S{fil_first_layer_flow}{endif}
```
in "after layer change gcode":
```
{if layer_num == 1 && exists(fil_first_layer_flow)}M220 S100{endif}
```
in "between extrusion role change gcode":
```
{if exists(fil_top_flow)}
{if extrusion_role=~/TopSolidInfill/}M220 S{fil_top_flow}
{elsif last_extrusion_role=~/TopSolidInfill/};M220 S100
{endif}
{endif}
```
(i didn't test it but it should be more or elss somethign like that)

courtross2010, 2022-08-31:
Cool, thanks for this info.  I'll give it a go.  Still think it would be good to have it in the UI settings but this will work in the meantime.  Thanks for all your hard work on this.

ph0tic, 2023-07-01:
+1 this, flow is definitely something im having to tweak on a filament by filament basis