supermerill/SuperSlicer #3139: Add Profile for Tronxy XY-2 PRO and Tronxy X5SA PRO

paulo9711, 2022-08-27:
Hi, how are you?
Can you add the profile of Tronxy XY-2 PRO and X5SA PRO?
Thank you!

Best regards.

neophyl, 2022-08-27:
Profiles are generally user submitted.  
Profiles are pulled from https://github.com/slic3r/slic3r-profiles and also the prusa list at https://github.com/prusa3d/PrusaSlicer-settings/tree/master/live (at least I think the prusa list ones are pulled in too)

So until someone with one of those printers does so there isn't much to be done.  
As you have one why dont you start with a similar printer and modify it and check it works and then submit one and help out the community.

paulo9711, 2022-09-05:
I have Tronxy profiles and I'm making adjustments. My friend has the GTMax3D. I uploaded the profiles here and couldn't. Can you (@neophyl)or can anyone tell me how I can do it?

supermerill, 2022-09-09:
fork https://github.com/slic3r/slic3r-profiles
`git clone https://github.com/paulo9711/slic3r-profiles.git`
modify the files in your computer.
`git commit -m "updates for Tronxy`
`git push`
Then click on the "git pull request" button that should appear on https://github.com/slic3r/slic3r-profiles

muehle79, 2023-01-21:
@paulo9711 did you try to upload the tronxy x5sa profile? or is it possible to send it to me?