supermerill/SuperSlicer #3141: Infill lines printing in midair/not connecting to infill beneath it

germgoatz, 2022-08-27:
### What happened?

Made a small 3d rectangle in Fusion360, sliced with 4% cubic infill, and it does this. Seems like the infill is not cubic and just random lines. I don't think this happened in earlier versions.

### Project file & How to reproduce

[infillbad.zip](https://github.com/supermerill/SuperSlicer/files/9437570/infillbad.zip)
basically explained how to reproduce lmk if i need to specify more

### Version

2.5.59

### Operating system

Windows 11

### Printer model

Ender 3

neophyl, 2022-08-27:
The reason its doing that is you have the infill angle increment value set to 270 degrees.  That means for every layer it rotates the infill pattern a further 270 degrees from what it normally is.  
Infill>advanced>Angle>Increment.  Set it to zero.  I'm guessing you meant to change the Fill value instead.

It happens in all the previous versions that support that setting.  I asked for it to be added :)

germgoatz, 2022-08-27:
> The reason its doing that is you have the infill angle increment value set to 270 degrees. That means for every layer it rotates the infill pattern a further 270 degrees from what it normally is. [...]

Oh wow I can't believe I didn't think of that. Thank u

germgoatz, 2022-08-27:
@neophyl Then is there some way to keep the top and bottom layers at that increment still without affecting the sparse infill? I was using it to keep the top and bottom layers aligned in the same direction.

closed 2022-08-27 by germgoatz

neophyl, 2022-08-27:
Just use a height range modifier to put it back to zero for the infill.  Look in the attached project. 
[infill_good_neo.zip](https://github.com/supermerill/SuperSlicer/files/9438538/infill_good_neo.zip)

germgoatz, 2022-08-27:
> Just use a height range modifier to put it back to zero for the infill. Look in the attached project. [infill_good_neo.zip](https://github.com/supermerill/SuperSlicer/files/9438538/infill_good_neo.zip)

Should have thought of that too. Thanks for your help :)