supermerill/SuperSlicer #3142: Bridge infill merged inside overhang perimeters

celtare21, 2022-08-27:
### What happened?

The bridge infill being at a higher layer height is getting merged with the previous layer's overhang perimeters:
[image]
[image]

### Project file & How to reproduce

Project file:
[Overhang_Angle_Test_55-85.zip](https://github.com/supermerill/SuperSlicer/files/9437726/Overhang_Angle_Test_55-85.zip)
 

### Version

2.4.58.4

### Operating system

Windows 11

### Printer model

custom ender 5 pro

CreativeFlug, 2022-08-28:
I have this issue on the bottom of the 3DBenchy, so the text becomes unreadable, but It's only on a 0.6 mm nozzle though. I haven't found any proper solution to it yet, the closest I have gotten is to change "Bridge flow baseline" to "keep current flow" but bridging lines becomes extremely thin when you do that

supermerill, 2022-08-31:
That's because the bridges have the same width as their height.
You can modify the way it's laydown in the bridge flow settings. Select "layer height" if you want to be sure it won't ever overextrude, but their quality may suffer.
[image]