supermerill/SuperSlicer #3146: Missing code signature in macos intel v 2.4.58.4

suschizu, 2022-08-28:
### What happened?

EXPECTED: Remi Durand (LDTLFRHP3G)
`SuperSlicer_2.4.58.4_macos_intel_220811_folder/SuperSlicer.app: code object is not signed at all`
app won't launch without permissions 

…the arm version is signed correctly

### Project file & How to reproduce

`codesign -dv --verbose=4 /Path/To/Application.app`

### Version

2.4.58.4

### Operating system

macOS

supermerill, 2022-08-29:
I tried to sign the intel version, and merged it with the arm to create an universal app (the arm is in fact the universal one) but I had a report that it doesn't run on intel.
I don't really know what's the problem, so I publish that one as a fallback plan.
Maybe I'll succeed next time?

VaughnR60, 2023-01-05:
Dont seem to be able to get Super Slicer to run on Big Sur (2014 iMac with Intel i5 processor) ... not smart enough to know why but ended up here trying to find an answer :). 

Has this been resolved or does the app still require signing?