supermerill/SuperSlicer #3148: Area threshold for ironing feature

craxoor, 2022-08-29:
Hey!

I thought it would be cool to have an area threshold for the ironing feature like for the massive infill. This would prevent the ironing of very small areas which leads to a very poor surface finish.

Currently I define a modifier cube to deactivate ironing in certain areas.

Keep up the good work!
Greetings, Greg

--------------------

Current state of ironing:
[image]

Ironing only on large surfaces:
[image]

TerrorzwergOlli, 2022-08-30:
That would be very nice!

supermerill, 2022-08-31:
small = small width, but can be very long ?

craxoor, 2022-08-31:
You are right. Using an area threshold would not catch very long or wide areas. Min width/height could work better in this regard.

ChaosBlades, 2023-05-01:
I had to use height range modifier to resolve my issue. Regardless of a feature to adjust the size of surfaces to be ironed the default is incredibly small like in my example. I would not even consider my example a top layer.
[image]
[image]