supermerill/SuperSlicer #3150: Crash app if change the 'Base Layer height' to a value other than in the 'Print Settings'

cypeks, 2022-08-30:
### Version
Version SuperSlicer-ubuntu_18.04-2.4.58.4.AppImage and SuperSlicer-ubuntu_18.04-2.5.59.0.AppImage

### Operating system type + version
Debian GNU/Linux testing bookworm (x86-64)

### Behavior
* _Describe the problem_
  * The application crashes and quits unexpectedly.
* _Steps needed to reproduce the problem_
  * Add two or more STL objects.
  * In one of the objects, change### Version
Version SuperSlicer-ubuntu_18.04-2.4.58.4.AppImage and SuperSlicer-ubuntu_18.04-2.5.59.0.AppImage

### Operating system type + version
Debian GNU/Linux testing bookworm (x86-64)

### Behavior
* _Describe the problem_
  * The application crashes and quits unexpectedly.
* _Steps needed to reproduce the problem_
  * Add two or more STL objects.
  * In one of the objects, change the 'Base Layer height' to a value other than in the 'Print Settings'.
    * Right-click on the object
    * Select 'Perimeters_Shell'
    * Change 'Base Layer height'
  * Press 'Slice now'
* _Expected Results_
  * g-code
* _Actual Results_
  * Crash application.

#### Project File (.3MF) where problem occurs

[crash.zip](https://github.com/supermerill/SuperSlicer/files/9452260/crash.zip)
 the 'Base Layer height' to a value other than in the 'Print Settings'.
    * Right-click on the object
    * Select 'Perimeters_Shell'
    * Change 'Base Layer height'
  * Press 'Slice now'
* _Expected Results_
  * g-code
* _Actual Results_
  * Crash application.

#### Project File (.3MF) where problem occurs

[crash.zip](https://github.com/supermerill/SuperSlicer/files/9452260/crash.zip)

jhlchu, 2022-08-31:
Seems similar to #3115? Crash in 2.4.59 Win, 2.5.59 Pre-release but fixed in 2.5.59 Nightly Win.

supermerill, 2022-08-31:
probably.