supermerill/SuperSlicer #3152: External perimeters first

Slatkey, 2022-08-30:
Hello. Is it possible to add the function "external perimeters first" parameter - "only areas without overhangs"? This will improve the quality of the exterior walls and prevent overhangs from falling.

supermerill, 2022-08-30:
it's possible.

Slatkey, 2022-08-30:
> it's possible.

Thanks! I'll be looking forward to it.

3Dxdream, 2022-09-01:
Excellent! Maybe with an angle threshold, if possible.
Thanks!

supermerill, 2022-09-01:
> Excellent! Maybe with an angle threshold, if possible.

I was thinking about reusing the "overhang speed/fan" threshold.

3Dxdream, 2022-09-02:
Do you mean the Perimeter&Shell/Overhangs section, with the "Bridge speed and fan" and "Bridge flow" thresholds?
That sounds great!
An additional checkbox at the right side like "no perimeter first", with the threshold edit box mouseover info "Minimum unsupported width for an extrusion to deactivate the perimeter first option if enabled."