supermerill/SuperSlicer #3153: Seam-sphere seams seem not saved with project

jhlchu, 2022-08-30:
### What happened?

Seam positions added via seam-sphere are not loaded when project is re-opened. Seam positions from seam-sphere only apply in same session seam-sphere is added. Re-opening same project shows seam-sphere object still exists but seams revert back to what is specified in "Print Settings" instead. Seam-sphere seams seemingly seem not saved.

### Project file & How to reproduce

1. Create a project and add an *.stl.
2. Right-click *.stl and select "Add seam position" via context menu.
3. Place seam-sphere to where seams are desired. View sliced gcode preview to confirm.
4. Save project and close superslicer.
5. Re-open superslicer and open project.
6. Slice.
7. Seams are now ignoring seam-sphere location and default to whatever is set in "Print Settings"

### Version

2.5.59 Nightly

### Operating system

Win 10 Enterprise

supermerill, 2022-08-30:
can't reproduce.
If you have a project that has an ignored seam-sphere on load+slice, give it to me.

jhlchu, 2022-08-31:
## Project File
[Seam.zip](https://github.com/supermerill/SuperSlicer/files/9458620/Seam.zip)

## Seam sphere position
[image]

## Correct seam in same session
[image]

## New session, same file
[image]

supermerill, 2022-08-31:
I can't extract your zip file, maybe there was an upload error?

jhlchu, 2022-09-01:
This one is just the extension changed *.3mf to *.zip instead of compressing the *.3mf. 
## Project File
[90Deg.zip](https://github.com/supermerill/SuperSlicer/files/9467180/90Deg.zip)

supermerill, 2022-09-06:
funny. I found this bug last February, and I fixed it. Now the root cause (loading modifier position) seems "fixed" and I need to remove the old fix.

VanessaE, 2023-07-19:
This occurs in the Ubuntu 18.04 Appimage build also (2.5.59.0).

VanessaE, 2023-07-19:
In fact, it happens with 2.5.59.2 as well.  Exactly as described by the parent. 
 I've attached a project file demonstrating it:

[Pneumatic Actuator (21828, plunger, with support structure).3mf.zip](https://github.com/supermerill/SuperSlicer/files/12094032/Pneumatic.Actuator.21828.plunger.with.support.structure.3mf.zip)

VanessaE, 2023-07-19:
In this project, a correctly-placed seam would be dead center at the back of the support tower structure as well as the back of the round shaft of the plunger object, and that's more or less what I get if I place a seam object appropriately and slice (it's not entirely correct but it's close enough).

However, if I save, restart, reload, and slice again, the plunger object is still correct, but the support structure has the seam in the wrong place.  I wonder if it's because the support structure is not an independent object, but instead was loaded onto the plunger object as a "part"?

Note that I even have the global seam position set to "Rear" in this project, so there are two things telling the slicer to put the seam on the back side.

closed 2024-02-11 by supermerill