supermerill/SuperSlicer #3156: Arachne settings disabled when installed using chocolatey

MrBlackmidi, 2022-08-31:
### What happened?

When i tried to install using chocolately, the arachne setting are either disabled or missing
[image]

### Project file & How to reproduce

Installed using chocolatey

### Version

2.4.58.4

### Operating system

Windows 11

### Printer model

All

MrBlackmidi, 2022-08-31:
[image]
img incorrectly cropped

jhlchu, 2022-08-31:
Arachne available in Pre-release SuperSlicer 2.5.59.0, not Latest 2.4.58 (you have this one). If you want arachne version, you can get it from releases page.

closed 2022-10-23 by MrBlackmidi