supermerill/SuperSlicer #3157: Curly brace added in filename

Stooovie, 2022-08-31:
### What happened?

SS adds a single closing curly brace into the filename, between the layer height number and "mm". This caused issue with the moonraker-telegram plugin.
[image]

### Project file & How to reproduce

Slice a file and send into a printer.

### Version

2.5.59.0

### Operating system

Macos 12.5

### Printer model

Ender 3

neophyl, 2022-08-31:
Check what you have defined in your print profile for the name field.  If you could attach your zipped up 3mf project file as requested when opening an issue we could take a look and check.

Stooovie, 2022-08-31:
Ah,, it's an Expert setting! This is what I have (I don't recall changing or even seeing this before):
`{input_filename_base}_{print_time}_{digits(layer_height,1,2)}}mm_{temperature[0]}C_{filament_type[0]}_{printer_model}.gcode`

I see an extra closing curly brace there, maybe a small bug? Issue solved though, thank you.

supermerill, 2022-08-31:
creality profile error, yes.
Thanks for the report.

closed 2024-02-11 by supermerill