supermerill/SuperSlicer #3161: Retraction when Tool is disabled -> Minimum Retraction

Ch4rlesB, 2022-08-31:
### What happened?

**Behavior**
I am using a corexy on Klipper. After a tool-change, filament just oozes out due to pressure inside the hotend. Theoretically, this setting should be able to retract the filament out of the hot zone and detract it back down after that tool is activated.

Retraction when Tool is disabled -> Minimum Retraction -> setting has no effect in generated code

My understanding of this setting is the length of filament retracted before calling the "Tool change Gcode".
No matter what values I put in there, I can't find them in the generated code

Basically, when the tool is parked, there is no retraction done beforehand. I can enter whatever number and it will never appear in the Gcode.

### Project file & How to reproduce

[Minimum Retraction.zip](https://github.com/supermerill/SuperSlicer/files/9464804/Minimum.Retraction.zip)

I have tried with a value of 12345. 
When searching in the Gcode file, that distance was only found in the parameters listed at the end of the file.

### Version

2.5.59

### Operating system

Windows 11

### Printer model

Dual Head SnakeOil-XY

supermerill, 2022-08-31:
Currently it only works if you activate the "wipe when retracting" and have no wipe tower (like in prusalicer, don't tell me why).

I guess It's something that can be improved, or at least I should update the tooltip.

Ch4rlesB, 2022-08-31:
Shouldn't the wipe tower be there to catch IF there is oozing from the nozzle due to a non-perfect retract tuning?
A combination of both settings is kind of needed to have a decent-looking IDEX print. Cura works like that and it seems to give good results.