supermerill/SuperSlicer #3172: Slicing weird holes

Assholer91, 2022-09-01:
### What happened?

Trying to print a coffe machine tray and it has couple of coffe grain shaped holes and two larger oval holes for water drainage on the bottom. When I slice the model it makes a mess where the holes are suposed to be. It is the same in Prusa slicer. It was sliced normally in Cura when I printed it the first time. 

[image]

[image]

I tried with diferent orientations and settings, with no success.

### Project file & How to reproduce

[TXXPRO_podstavek_za_kafetiero_v2.1SS.zip](https://github.com/supermerill/SuperSlicer/files/9471736/TXXPRO_podstavek_za_kafetiero_v2.1SS.zip)

### Version

2.4.58.4

### Operating system

Win 10

### Printer model

Tronxy X5SA Pro

3Dxdream, 2022-09-02:
It shows "98820 open edges" in your stl-model.
But you can press the blue warning sign symbol and repair it. 
Then it works!

Assholer91, 2022-09-02:
Now this was simple. 

Thank you for solution, I will look up why this happened with the model to prevent it next time.

closed 2022-09-02 by Assholer91