supermerill/SuperSlicer #3175: [FR] Expressions & Variables in numeric input fields

jhlchu, 2022-09-04:
**Is your feature request related to a problem? Please describe.**
I use the same Print Setting profile for printing small and large objects, but sometimes the speed (using autospeed) blitzes some of the layers on small objects, requiring some manual adjustment to the profile. Want to be able to selectively override autospeed or other values with expressions like `min` or `max` and use printer variables in them.

**Describe the solution you'd like**
Being able to use basic expressions and printer variables in numeric input fields.

**Describe how it would work**
- "Print Settings" > Speed > External: `min(15, InternalPerimeter.speed)`
- "Print Settings" > Width > Perimeter: `max(0.4, LayerHeight * 0.8)`.

**Describe alternatives you've considered**
Making manual adjustments then slicing, to see.

supermerill, 2022-09-06:
too problematic with cyclic references.

supermerill, 2022-09-06:
That motivated me a bit to test & debug the scripted widget.
starting from next nightly, add that in your prin.as:
```
float s_ext_peri_speed_get(bool &out isper)
{
	isper = is_percent("external_perimeter_speed");
	float fl = get_float("external_perimeter_speed");
	if(isper || fl < 15) {
		ask_for_refresh();
		return 15;
	}
	return fl;
}

void s_ext_peri_speed_set(float val, bool isper)
{
	if(!isper){
		if( 15 > val) {
			val =15;
		}
		set_float("external_perimeter_speed", val);
	}
}

float s_peri_extr_width_get()
{
	float fl = get_computed_float("perimeter_extrusion_width");
	float layer_height = get_float("layer_height");
	float val = fl;
	if(layer_height * 0.8 < val) {
		val = layer_height * 0.8;
	}
	if( 0.4 > val) {
		val = 0.4;
	}
	if(val != fl){
		ask_for_refresh();
		return val;
	}
	return fl;
}

void s_peri_extr_width_set(float val)
{
	float layer_height = get_float("layer_height");
	if(layer_height * 0.8 < val) {
		val = layer_height * 0.8;
	}
	if( 0.4 > val) {
		val = 0.4;
	}
	set_float("perimeter_extrusion_width", val);
}
```
and in your print.ui, replace `setting:width$4:external_perimeter_speed` by `setting:script:float_or_percent:depends$external_perimeter_speed:tags$Simple$Advanced$Expert$SuSi:label$My external speed:tooltip$issue 3175:s_ext_peri_speed`
and 
```
line:perimeter
	setting:sidetext_width$10:label$width:perimeter_extrusion_width
	setting:sidetext_width$10:label_width$15:label$spacing:perimeter_extrusion_spacing
end_line
```
by `setting:script:float:depends$perimeter_extrusion_width:tags$Simple$Advanced$Expert$SuSi:label$My perimeter width:tooltip$issue 3175:s_peri_extr_width`

closed 2024-02-11 by supermerill