supermerill/SuperSlicer #3176: Extruder page management broken in Master branch

nanoplane, 2022-09-04:
### What happened?

if you have a printer settings tab with multiple extruders defined, multiple extruder pages are defined and numbered sequentially from "1".  save this printer settings.
if you return to the printer page and add/remove extruders, the Extruder pages on the left are mangled. see attached image

This first screenshot is before modifying the count
[image]

This second screen shot is after modifying the count (add one extruder).
[image]

### Project file & How to reproduce

no project file required.... 
1. create new printer setting
2. add extra extruders
3. save printer setting
4. add or remove extra extruders..
5. see the result.

### Version

2.4.58.4

### Operating system

MacOS 15

### Printer model

any

nanoplane, 2022-09-04:
The issue is in the Tab.cpp file, in the function TabPrinter::build_unregular_pages, line 3149.
at this point, if there are multiple extruders and the count is changing and it's not from_initial_build, then the value of n_before_extruders is incorrect.

This can be corrected by inserting the following before line 3149:

    if (m_has_single_extruder_MM_page) {
        n_before_extruders++;
    }

nanoplane, 2022-09-05:
Here's another sample. Created by having "Single Extruder MultiMaterial" checked and clicking the up and down arrows of the "Extruders" field multiple times.

Before:
[image]

After:
[image]

nanoplane, 2022-09-05:
The Single extruder MM setup page issue is another bug...

supermerill, 2022-09-09:
I can see the problem with the single extruder mmu.
But I don't see any problem with your second screenshot. please enter in red the problematic thing.

nanoplane, 2022-09-09:
Just look at the ordering of the Extruders.. and the fact that there are two entries for Extruder 3

nanoplane, 2022-09-09:
I've created a fix for this at nanoplane/SuperSlicerMx::extruder_tab_fix.. two functions modified.