supermerill/SuperSlicer #3183: Combine seam sphere with other seam settings.

cfytrok, 2022-09-06:
Could you make a detailed description of how the seam sphere works? Is it possible to combine seam sphere with other seam position settings? With seam painting? With Height range modifier?

jhlchu, 2022-09-07:
> Is it possible to combine seam sphere with other seam position settings?

No. Seam sphere seams seem to override seam Modifiers and "Print Settings" seams. Image below I set seam to corner in "Print Settings". The sphere is seam sphere, white dots are seams. Seam sphere seam locations appear to be a projection from the origin of the sphere onto the closest surface on the object, usually one per layer.
[image]
[image]

> With seam painting?

Seam painting can override seam sphere. Order of precedence is as such: Seam painting > Seam sphere > Seam Modifier & Settings
[image]
[image]

> With Height range modifier?

Is there a seam modifier for height range?

### This doesn't seem to be an "Issue". "[Discussions](https://github.com/supermerill/SuperSlicer/discussions)" is probably the appropriate place for this topic.

supermerill, 2022-09-07:
The center of the sphere attract the seam.
If you put more than one, the one used is the one that is the nearest in the z direction.