supermerill/SuperSlicer #3185: Internal brim overlapping same layer

panik988, 2022-09-06:
### What happened?

The internal brim is printed 2 times overlapping the same layer

### Project file & How to reproduce

[error_interior_brim.zip](https://github.com/supermerill/SuperSlicer/files/9497684/error_interior_brim.zip)

### Version

2.5.59

### Operating system

window 11

### Printer model

vzbot 330

tinyfluffs, 2022-12-03:
Not sure if this is the same issue, but I've also noticed a double brim on some parts. Running version 2.4.58.5.

Project files & Slicer config: [bmg_bug.zip](https://github.com/supermerill/SuperSlicer/files/10146238/bmg_bug.zip)

[image]

lusanagi, 2023-10-30:
I'm using version 2.5.59.2 and still get this issue :/
Gcode looks good, but when printing it has the described behavior, which causes the brim to be lifted from the heat plate (only happens with brim, brim ears work correctly).
I do see that even though I have my first layer height set at 0.44 (0.6 nozzle) in the gcode it is set as 0.4375, not sure if that is relevant
[image]

Running TwoTrees SP5 with klipper (pretty much a Voron Trident); For now I'm running Orca with brimmed prints or if I need tree supports, but besides not having fine tuning like susl, I just realized that estimation is way off than real print time (some times by 2x), and they already stated they won't fix

supermerill, 2023-12-29:
still can't reproduce in latest 2.5.59.5 release
If you can reproduce it, please put here the problematic project.

lusanagi, 2024-01-14:
All projects give me this issue (I even have a 0.2 layer profile with a first layer of 0.24 and have the same issue); I found that as a workaround I can put the first layer as 0.4 (and 0.2) and the gcode will have the correct height 

I just saved a project with my 0.4 profile and attached here
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/13933466/Shape-Box.zip)

lusanagi, 2024-01-15:
@supermerill when I mentioned the issue I was on 2.5.59.2, but the project I uploaded yesterday was saved in 2.5.59.6

supermerill, 2024-01-15:
@lusanagi 
can't reproduce with fresh 2.5.59.2, nor 2.5.59.6 on win10.
[image]

I miss something. Maybe I don't look at the right thing? I checked it via the gcode viewer, and the brim is printed only one time.
Maybe there is something else that mess it. Can you test from an empty configuration directory?

lusanagi, 2024-01-15:
@supermerill I'll try that when I get to my computer, but I can tell you that yeah, on gcode viewer it looks good, but if you check the gcode with a text editor you can see that Z height is reduced by 0.025 (0.4375)
I'm guessing this is the reason the brim is lifted while printing (which is the issue I first mentioned), but I can't say for sure

supermerill, 2024-01-15:
your z_step is 0.00625
your first layer height is 0.44
0.44/0.00625 = 70.4
rounded to 70 : 70*0.00625 = 0.4375
so the first layer height is set to 0.4375
then after that, the slicing and all the stuff is done.

lusanagi, 2024-01-16:
Well, that's something I didn't know about and don't know how I got that value 🤦
I just set the z_step to 0.005 and get the correct first layer

Thanks for your help!

closed 2024-02-11 by supermerill