supermerill/SuperSlicer #3191: Brim is printed for all objects first, then objects are printed

dewi-ny-je, 2022-09-08:
### What happened?

When I have multiple objects SS first prints all brims, then it starts with the objects. This results in the brim not being perfectly connected to each object, since it has not been printed as a continuous way with the outermost perimeter. I find objects curling up while the brim stays flat!

Each brim should be printed not at the beginning but together with the respective object.

### Project file & How to reproduce

[test brim.zip](https://github.com/supermerill/SuperSlicer/files/9530427/test.brim.zip)

### Version

2.4.58.3

### Operating system

Windows 10

### Printer model

HEVO with PIF profile from AndrewEllis

jhlchu, 2022-09-09:
Brim isn't meant to be perfectly connected to the object or else it becomes too difficult to remove.
Does reducing your "Print Settings" > "Brim Separation Gap" help with the issue?

jhlchu, 2022-09-09:
Going off you suggestion, I decided to try a negative "Brim Separation Gap". Indeed you can, but I will overlap parts of your first layer depending on how much you set it.
I see your point. I've only been printing PA and PC with brim with 1 object on the platter, so I haven't ran into that issue yet. I guess another point for your suggestion is materials with a tendency to ooze (TPU, PETG, PA). Printing the object with it's corresponding brim reduces the ooze during traveling from brim to brim then back to object.

Point 4 is definitely a bug. I just tried with 3 objects on the platter. The order in which brim is printed and the order which first layer is printing is mis-matched. It went brim 3, brim 1, brim 2, object 1, object 2, object 3 for me (2.5.59 05585fd).

The only workaround I found for this so far is to make "Brim Width" large enough that they overlap so that it prints as one piece and reduces (but not eliminates) the cool down.

dewi-ny-je, 2022-09-09:
Thanks for the checks.

Overall, I think that the "reproducibility" and "stronger bond, if desired" advantages justify quite well changing the sequence of the brims!

As said, nothing would be taken away to those desiring a weaker bond!

dewi-ny-je, 2022-09-09:
Which would be the file to lok at if I wanted to try changing the sequence? I don't think my programming skills allow it, but I can give it a look. I only don't know where to look.

jhlchu, 2022-09-09:
Take a crack at it.
Brim is generated in https://github.com/supermerill/SuperSlicer/blob/eab5d801234a595fdcc28019e2afb3b4c5fc86fa/src/libslic3r/Brim.cpp
Called in https://github.com/supermerill/SuperSlicer/blob/eab5d801234a595fdcc28019e2afb3b4c5fc86fa/src/libslic3r/Print.cpp#L1138

dewi-ny-je, 2022-09-09:
> Take a crack at it. Brim is generated in https://github.com/supermerill/SuperSlicer/blob/eab5d801234a595fdcc28019e2afb3b4c5fc86fa/src/libslic3r/Brim.cpp Called in
> 
> https://github.com/supermerill/SuperSlicer/blob/eab5d801234a595fdcc28019e2afb3b4c5fc86fa/src/libslic3r/Print.cpp#L1138

Two lines below
https://github.com/supermerill/SuperSlicer/blob/eab5d801234a595fdcc28019e2afb3b4c5fc86fa/src/libslic3r/Print.cpp#L1140
Instead of

`                if (config().complete_objects && !config().complete_objects_one_brim) {
`
It should be 

`                if (!config().complete_objects_one_brim) {`

But it's not perl anymore :) so it has to be compiled.
How does that work?

Can someone with a working development environment compile it so that I'll do the testing?
Thanks!

supermerill, 2022-09-09:
Brims are printed first, because they can merge together.
I can makes them to be printed sequentially (i made something for sequential printing). But you may have brims that are cut like that:
[image]

dewi-ny-je, 2022-09-09:
@supermerill  Thanks for thinking about my suggestion.

> Brims are printed first, because they can merge together. I can makes them to be printed sequentially (i made something for sequential printing). But you may have brims that are cut like that: [image]

But if the option "individual objects" prints them before each object, shouldn't the problem be there as well?

Anyway I see your point and unless a more complex algorithm is developed which generates them first, but prints them somehow before each object as I suggested, keeping the current default is the best choice.

Still, an option to print brims before each object for more advanced users which are aware of the consequences and are therefore going to manually leave enough distance between objects, would bring all the advantages I listed.

It seems a useful feature, even if I agree it should not be the default.

dewi-ny-je, 2022-09-09:
It's also an enhancement which requires extremely little effort, since besides the GUI only a couple lines need to be changed.

supermerill, 2022-09-09:
> It's also an enhancement which requires extremely little effort, since besides the GUI only a couple lines need to be changed.

lol
If so, make it and submit a pull request