supermerill/SuperSlicer #3193: Strange perimeters whit MMU paint

efradc, 2022-09-09:
### What happened?

hi, Ihave these strange perimeter problems with SS, with PS it doesn't happen to me...
Also try to export 3mf from PS and open with SS, same problems... maybe a bug?

I have tried with 2.4 and 2.5.

PS slice:
[image]

SS Slice:
[image]

Thanks!

### Project file & How to reproduce

Import painted 3mf or paint a new one and slice

### Version

2.4.58.4

### Operating system

windows 10

supermerill, 2022-09-09:
I just fixed it
#2639

efradc, 2022-09-09:
thanks, sorry. I was looking if it was solved but I didn't find

closed 2022-09-09 by efradc

supermerill, 2022-09-09:
forgot to tell you about how to solve it:
If you increase the resolution (reduce the number, to 0), then it can apply the paint on the surface again.

efradc, 2022-09-20:
After trying many things I have seen today the answer hahaha now it is perfect, thank you very much for the work you do