supermerill/SuperSlicer #3196: latest nighty_dev crashes when selecting printer with multiple extruders

moo00se, 2022-09-10:
### What happened?

latest nighty_dev (uploaded 16 hours ago) crashes when selecting a printer with multiple extruders

When opening SS, immediately selecting the printer with multiple extruders, SS crashes. 

### Project file & How to reproduce

open SS, select a printer with multiple extruders.

### Version

latest nightly_dev

### Operating system

windows 64

### Printer model

voron switchwire

meoudon, 2022-09-11:
[image]
I'm use nightly too have no problem.

moo00se, 2022-09-11:
Interesting, ill have to dig a little deeper on why. Thanks.

supermerill, 2022-09-12:
give us a zip of your configuration directory (from help -> show conf dir), so I can replicate.

moo00se, 2022-09-14:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9568097/SuperSlicer.zip)
sorry for the delay, I went back a nightly_dev. 
I tested the latest nightly_dev and it does not crash as the one before did. So just something with that one specifically and my setup.