supermerill/SuperSlicer #3197: Wierd inside wall perimeters when in wall is hole

technowsky, 2022-09-10:
### Version
2.5.59.0

### Operating system type + version
Windows 10, version 21H2

### Behavior
Inside wall perimeters are slightly extended when in wall is a hole. I checked the same stl in PrusaSlicer 2.5.0 alpha3 and issue doesn't appear. In Cura 5.1 problems also appear. I don't know if my settings are causing an issue or if this is an internal problem in slicers.

SuperSlicer:
[image]

PrusaSlicer:
[image]

Cura:
[image]

#### Project File (.3MF) where problem occurs
[model.zip](https://github.com/supermerill/SuperSlicer/files/9540915/model.zip)

neophyl, 2022-09-11:
You have vertical hole compensation enabled.  Slicing>Modifying slices>Vertical Hole Compensation.  Set it to zero and it goes away.

There are far more options in SuSi and you really really have to learn what they do.  I dont know why Cura is also doing it as I dont use Cura but Id suspect theres a setting buried in there too.  PS doesnt have many of these types of settings.  If they added them they would get far far more customer issues as these are for more advanced users who want nad need more control.  With great power comes yadda yadda yadda :)

supermerill, 2022-09-12:
also, prusa seems to use much higher widths (or the model is scaled down)

with at least 0.46 width for superslicer, you can get backs what you want.

Explanation: there isn't enough width below the "yellow diamond" for a normal extrusion, and there isn't anough width between them for an extrusion loop, hence why everything is filled with gapfill.