supermerill/SuperSlicer #3200: Generating weird bridges inside hollow model

Arakon, 2022-09-11:
### What happened?

When slicing some models, without infill (or even with), a lot of floating internal bridges and solid infill perimeters are generated.
In comparison, the model remains hollow in Cura.

### Project file & How to reproduce

[Cali-Dragon_v1.zip](https://github.com/supermerill/SuperSlicer/files/9541997/Cali-Dragon_v1.zip)

### Version

2.5.59.0

### Operating system

Windows 10

### Printer model

VZBot

neophyl, 2022-09-11:
Thats because you have supporting dense layer enabled.  That is what is generating the blue bridges inside, as that setting is there to put a initial layer below top layers that are generated above.  
The extra infill is because you also have ensure vertical shell thickness enabled.  Although even if you turn that off it will still do some unless you tweak the No solid infill over parameter too.  

That model though doesn't look like it will print properly without infill as there are too many unsupported areas on layers above.

You also need external support under the chin.

jhlchu, 2022-09-11:
It's in `Print Settings>Infill>Reducing Print Time>Supporting Dense Layer`. Disable to remove them.