supermerill/SuperSlicer #3201: Feature request: Make Overhangs Printable (Auto Fillets)

inspectionsbybob, 2022-09-11:
2.4.58.4
_Version of SuperSlicer used goes here (help->about)_
Win10
_What OS are you using, and state any version #s_

Make Overhanges Printable
On many models, there can be a "lip" or "Shelf" with a 90deg edge that requires support.  Sometimes those supports come from the Baseplate, which can be many, many inches away.  Would like to see a feature similar to Cura's that puts Fillets on the underside to "Make the Overhang Printable."

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2022-09-12:
Do you have a link to cura's feature, or at least some pictures?

inspectionsbybob, 2022-09-13:
Here you go... the effect is subtle, but very powerful

jhlchu, 2022-09-13:
Nothing is attached, there is no image if you meant there to be one. Reply in github.com instead of the email.

inspectionsbybob, 2022-09-13:
[Screenshots.zip](https://github.com/supermerill/SuperSlicer/files/9561204/Screenshots.zip)
Try this one

inspectionsbybob, 2022-09-13:
First two are raw STL so you can see the junction
Next is the unfixed Gcode
Next is the Modified Gcode, toggle between them you will see the fillet for the band is much more pronounced

supermerill, 2022-09-13:
understood.
Something to modify the geometry, to enforce a threshold for overhangs (but maybe not for detected bridges).
Can remove material if going from the layer 0 to the top, or can add material if going from the top to the bottom.

That's right?

inspectionsbybob, 2022-09-14:
I never thought of it removing material...

anyway, here is the github to it...

https://github.com/Ultimaker/Cura/issues/6966

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

theschles, 2022-10-19:
Duplicate of #2874  -- please close this or the other one?

inspectionsbybob, 2022-10-20:
Duplicate of 2874 -- please close this or the other one?

closed 2022-10-20 by inspectionsbybob