supermerill/SuperSlicer #3202: Feature Request: Simplify Model(s)

inspectionsbybob, 2022-09-11:
### Version
2.4.58.4

### Operating system type + version
Win10

### Behavior
"Simplify" overly complex Models.
Some Models are Built HUGE so the author can put in details, and then they are shrunk down to the appropriate size, but the details remain even though they are not printable as they are SMALLER than the Nozzle.  Similarly, when some models shrink, some gentle curves are now FLAT.

Desired is an option for "Simplify Model," which would use something like "Mesh Mixer" to remove/replace/simplify any element that is smaller than is printable or can be simplified because there are multiple planes side by side.  The option should calculate using the Nozzle Size and other parameters that determine the minimal printable object or if (optionable) Resolution is willing to be lost to reduce the complexity and thus print & rendering speed.

Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

jhlchu, 2022-09-11:
Would this replace the existing "Simply Model" feature in SS or augment it?

supermerill, 2022-09-12:
what is the problem the current "simplify model" button ?

inspectionsbybob, 2022-09-13:
i couldn't find it... had a professional model that was -way- over
detailed. Took forever to load... couldn't find the simplify button...
maybe it hadn't finished drawing the menus, it was that slow because if the
over detailed...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

supermerill, 2022-09-13:
right-clic on the object -> simplify model (~in the middle)

inspectionsbybob, 2022-10-03:
OK, close it

closed 2022-10-03 by inspectionsbybob