supermerill/SuperSlicer #3203: Newest Nightly Wont Load

The-Mechanix, 2022-09-12:
### What happened?

When I launch the newest nightly (September 9/2022), It will load the splash screen then not load the program. When I run the console this error occurs. 

[2022-09-10 22:40:29.510881] [0x00003aa4] [trace]   Initializing StaticPrintConfigs
[2022-09-10 22:40:30.007967] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\.2 quality speed .4 width PETG.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.023971] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\.2 quality speed .4width PLA.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.039978] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\.2 quality speed .4width.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.047977] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\.2 quality speed PLA.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.063982] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\.2 quality speed.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.071983] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\0.20mm NORMAL @CREALITY - Copy.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.087990] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\0.20mm NORMAL @CREALITY - PETG.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.095990] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\0.20mm NORMAL @CREALITY - tpu.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.111996] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\0.20mm NORMAL new.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.119996] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\ender 5 my pla.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.136017] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\My Settings.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed
[2022-09-10 22:40:30.152005] [0x00003aa4] [error]   Error in a preset file: The preset "C:\Users\Mike\AppData\Roaming\SuperSlicer\print\Tests.ini" contains the following incorrect keys: complete_objects_one_brim, which were removed

I have tried to find the culprit setting in my current config and cant figure out what is wrong to change it through the slicer its self. 

### Project file & How to reproduce

Download file, unzip, try to open .exe

### Version

2.5.59 Nightly

### Operating system

Winx64

### Printer model

Mercury One.1

coolio986, 2022-09-12:
I also have this issue. I have pulled the latest nightly. 

slic3r\GUI\Tab.cpp fails on "read access violation." Due to a nullptr on is_phony() / set_phony().

Tab::update_changed_ui()

at line 783, thrown at Config.hpp, ConfigOption line 412 and 413.

opt_key in this context is "complete_objects_one_brim"

system_preset->config.option does not contain this key, however selected_preset does

coolio986, 2022-09-12:
After removal of that key in the profile, software starts as expected.

Is there a mechanism in place already for profile setting detection and comparison? I'm a newb to SuperSlicer code, but not to c++ programming. 

I would expect that if a system preset doesn't exist in selected presets, it would remove / ignore useless keys from the ini.

But again, I'm a newb. Please correct me if I am wrong.

coolio986, 2022-09-12:
Attached is an zip/ini file for error reproduction.
[Qidi X-Plus.zip](https://github.com/supermerill/SuperSlicer/files/9550228/Qidi.X-Plus.zip)

supermerill, 2022-09-12:
I deleted the complete_objects_one_brim and kind of replaced it by brim_per_object
I'll add a check for the Tab.cpp "access violation" crash (even if I can't reproduce it, how to reproduce with your ini?)
I forgot to add it to the list of old unused settings to ignore. done.
@The-Mechanix : you said "Download file, unzip, try to open .exe" but I can't see any file.

coolio986, 2022-09-12:
To reproduce, copy the ini to appdata roaming, then simply try and open SS from nightly.

Without the debugger of course the app crashes. With the debugger you will get the error thrown to the debugging console.

Maybe the vector can be extended to add bounding checks, or maybe a simple null check could be used inline in Tab.hpp. 

My 2c

The-Mechanix, 2022-09-12:
> I deleted the complete_objects_one_brim and kind of replaced it by brim_per_object I'll add a check for the Tab.cpp "access violation" crash (even if I can't reproduce it, how to reproduce with your [...]

I should have been more precise, download this .zip https://github.com/supermerill/SuperSlicer/suites/8223623103/artifacts/358694453
unzip, then tried to run it, did not open. 

I ended up going into the config folder, making a backup, then manually removing complete_objects_one_brim from each configuration. That makes the program load as expected, and also loads without error. slicing produces no errors.

supermerill, 2022-09-12:
> To reproduce, copy the ini to appdata roaming, then simply try and open SS from nightly.

Reproduced & fixed.

> unzip, then tried to run it, did not open

It opens in my computer. The problem lie in your configuration. Can you zip it and send it to me, to be sure it's fixed (should be the same bug as coolio986)?

The-Mechanix, 2022-09-12:
> Reproduced & fixed.
>
>
> It opens in my computer. The problem lie in your configuration. Can you zip it and send it to me, to be sure it's fixed (should be the same bug as coolio986)?

Here are the configuration files that were causing
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9551778/SuperSlicer.zip)
 the issues for me before,

supermerill, 2022-09-12:
can open with the fix.
will be in next nightly.

coolio986, 2022-09-13:
Can confirm the nightly has fixed this issue.

closed 2024-02-11 by supermerill