supermerill/SuperSlicer #3204: Vase mode doesn't change to one vertical shell

JacTheMaker, 2022-09-12:
### Version
SS 2.5.59.0

### Operating system type + version
Windows 11 Pro

### Behavior
Check Vase Mode, Click Yes
Doesn't change number of vertical shells to 1.
Still Slices.

JacTheMaker, 2022-09-12:
[image]
[ribbon2.zip](https://github.com/supermerill/SuperSlicer/files/9544298/ribbon2.zip)

Just change file extension, not actually zipped

supermerill, 2022-09-12:
use classic slicing.
Also, the object ahs a section with walls thinner than your nozzle diameter, so it won't be able to vase-slice it there.

paulgiordano, 2023-03-11:
This code appears to cause a segfault at PrintConfig.cpp line 8058, 
`this->opt<ConfigOptionEnum<PerimeterGeneratorType>>("perimeter_generator", true)->value = PerimeterGeneratorType::Classic;`

closed 2024-02-11 by supermerill