supermerill/SuperSlicer #3206: Arachne still buggy in latest NB?

locki-cz, 2022-09-12:
### What happened?

Arachne still shows some artefacts in latesin NB.
[image]

### Project file & How to reproduce

It shows only on this top part ... on rest of model it works fine ... 

### Version

2.5.59.0 NB

### Operating system

Win 10

### Printer model

Voron 2.4 SB

zeroemso, 2022-09-12:
since it is arachne related i jump in here....found another reproducable bug.

at this part in the pic its just leaves a missing line on every 2nd layer. If i switch to classic its all normal (but Gap fill is used there)
[image]
[image]

supermerill, 2022-09-12:
@locki-cz your project?

locki-cz, 2022-09-12:
[Hera_CQR_motor_cover-arachne.zip](https://github.com/supermerill/SuperSlicer/files/9551775/Hera_CQR_motor_cover-arachne.zip)
Sorry i forgot! :(

ChaosBlades, 2022-09-13:
No Problems here on 2.5.590. I would revert to a stock slicing profile and try again.
[image]

ChaosBlades, 2022-09-13:
It is definitely enabled. Maybe it is either that I am using Linux or I have some setting that is causing the bug that is either disabled or changed in some way that is avoiding the issue.
[image]

[Hera_CQR_Stock_advanced-repaired-fixed.zip](https://github.com/supermerill/SuperSlicer/files/9555507/Hera_CQR_Stock_advanced-repaired-fixed.zip)

supermerill, 2022-09-13:
fixed by prusa with the full 2.5 merge.

locki-cz, 2022-09-13:
it looks same, that model is bad but why it doing this? :)

[image]

ChaosBlades, 2022-09-13:
Did a little more research and I found someone looking at the setting page for Arachne on SuperSlicer and it looks like it is bugged on Linux as I do not have that settings panel. I tried to remove both Stable and Arachne AppImage of SuperSlicer and reinstall. Also tried to create a new custom printer after that and the page still does not show up. I finally got it to show up by removing the .config folder. I don't know the situation on Windows/Mac but on Linux it appears anyone that updates and wants to use Arachne will need to start with a fresh install.
[image]

Once I got the Arachne settings to show up I was able to reproduce the bug.

supermerill, 2022-09-13:
> Did a little more research and I found someone looking at the setting page for Arachne on SuperSlicer

Check you're using the Configuration -> Preferences -> GUI -> Settings layout & colors -> STANDARD
and not the exemple.

ChaosBlades, 2022-09-13:
I am using Standard.
[image]
I fixed the problem by deleting the `ui_layout` folder in the .config directory. It does not appear so far that I have lost any important settings. Looks like the last time that folder was modified was the 15th of July.

pa0los, 2022-09-17:
I faced the same issue.
The workaround from @ChaosBlades worked for me as well.
Thanks!

closed 2024-02-11 by supermerill