supermerill/SuperSlicer #3209: Brim and Skirt Speed seems to be ignored

jscancella, 2022-09-14:
### What happened?

I change the `Brim & Skirt` under the `Print Settings` and `Speed`, yet it seems no matter what I put there it still just uses 50% of `Support Speed`.

### Project file & How to reproduce

[brim and skirt speed ignored example.zip](https://github.com/supermerill/SuperSlicer/files/9561850/brim.and.skirt.speed.ignored.example.zip)

Change the `Brim & Skirt` speed, slice, and see that it is still at 25mm/s in the gcode viewer. 

### Version

2.4.58.4

### Operating system

Windows 10

### Printer model

RatRig vcore3

jscancella, 2022-09-14:
Here you can see I set the `Brim & Skirt` speed to 100mm/s
[image]

But here you can see it is still doing only 25mm/s for the skirt
[image]

Kerman-ap, 2022-09-18:
pretty sure first layer speeds are used there. The speed you enter in will be used for the draft shield

Jens5395, 2022-12-03:
I just spent quite a while to try and figure out why my skirt or brim speed seems to be fixed. I can confirm that I too did not see any way to adjust skirt/brim speed. The setting seems to be ignored.

christi-ko, 2023-01-09:
Just to forward the information:
it was facing a similar issue, and there seems to be anotherone explained here:
https://forum.duet3d.com/topic/30613/superslicer-print-speed