supermerill/SuperSlicer #3210: Skirt goes under print in SuperSlicer

zogthegreat, 2022-09-14:
### What happened?

Hi everyone!

I'm having a strange problem with SuperSlicer v2.5.59. For some reason, the skirt that's supposed to go around the print is going under it:

[image]

Does anyone know what I'm doing wrong here?

Thanks!

### Project file & How to reproduce

[LINEAR_DUAL_Z_12h23m.zip](https://github.com/supermerill/SuperSlicer/files/9566016/LINEAR_DUAL_Z_12h23m.zip)

### Version

2.5.59

### Operating system

Windows 11

### Printer model

Ender 3 Pro

neophyl, 2022-09-14:
Firstly, attaching gcode is useless for debugging these issues.  To be certain we would need a copy of your saved project file.  

The usual reason for this sort of issue is that the part the skirt is going under is NOT flat to the bed but at a slight angle.  Usually using the lay part to bed tool will correct this.  Sometime though you have to flip it wildly and then flip it back to get it to sit flat.  
Hence the standard requirement for all issues of asking people to attach a zipped up project, file.

Pavulon87, 2022-09-15:
Your parts are not lying flat on the bed, so the skirt doesn't touch them.
First layer:
[image]
Second layer:
[image]

zogthegreat, 2022-09-15:
OK, thanks for the information. I just switched over to SuperSlicer so I'm on a learning curve here. I'll be more attentive to how things are on the plater in the future.