supermerill/SuperSlicer #3211: First Layer Height doesn't match settings?

inspectionsbybob, 2022-09-14:
### Version
2.4.58.4

### Operating system type + version
Win10

### Behavior
First layer Height doesn't match settings?
I have my "Base Layer" set to 0.15mm
I have my "First Layer Height" set to 75%

I EXPECTED my first layer to be .75 x .15 or 0.1125 or there about

It is ALWAYS MUCH more than that, as in 0.30-0.32 or thereabouts.

Also, default line widths are weird, but I assume those are from the 105% of .40 as first layers and others...

_Is this a new feature request?_
NO.... more a question as to what is it doing?

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/9569123/Shape-Box.zip)

inspectionsbybob, 2022-09-15:
Ok, I re-leveled the bed HOT,  waited and re-leveled it AGAIN, and then printed some 1-layer thick stripes that were SUPPOSED to be 0.28mm thick (my purge lines) set in my start gcode at Z 0.28 right after a G28.

They were 0.17mm thick.   The DISPLAY (and octoprint) said the Nozzle (z) was at 0.31mm.

I am really confused.  I am just going to set it up for a really good first layer regardless of what it says...

Might be something to clean up LATER....

BartVB, 2022-09-16:
This is not a SuperSlicer issue, it's a firmware issue.

When you are levelling your bed you probably use something like a piece of paper to guage if the nozzle is close to the bed. If you use the same piece of paper you can make your bed level to the XY-plane of your printer. The thickness of the paper is what's causing your issue here. You are not levelling the nozzle to 0mm from the bed, but rather something like 0.1mm from the bed. With Marlin you can use 'babystepping' to correct for that. There are plenty of youtube videos about that.

When you have fixed that Z-level offset then your first layer will be the height that you've set in SuperSlicer.

jrmann1999, 2023-02-08:
It appears the First Layer height calculation is based on Nozzle diameter, and not taking into account the actual desired layer height.

For example, I'm using a 0.6mm nozzle with 0.3 layer height.  First layer height is 75%, so As above I expected first layer to be 0.225 or roundabouts.  In the gcode it's 0.45, this is 75% of 0.6

Is this the intended behavior?  Shouldn't the calculation take into account the desired first layer height?