supermerill/SuperSlicer #3212: [FR] Setting for changing "Internal Bridge Infill" angle (Supporting Dense layer angle)

jhlchu, 2022-09-15:
**Is your feature request related to a problem? Please describe.**
The algorithm sometimes chooses the wrong direction for the internal bridge infill angle. It makes very long bridges when turning 90 deg will make short and straighter bridges. Turning the object on the platter fixes it for some layers but messes it up for the others.

### Bad 
[image]

### Good 
<img src="https://user-images.githubusercontent.com/5092505/190352267-57867faa-7690-4992-a78f-09c7d04c008e.jpg" 
 width=500 />

**Describe the solution you'd like**
Numeric input box next to `Print Settings > Infill > Advanced > Angle > Bridging` called `Internal Bridge Angle` or make `Angle > Bridging` setting also affect "Internal Bridge Infill". I can then use it in a modifier to change it where needed.

**Describe alternatives you've considered**
Rotating the object on the platter. This fixes some angles, makes other angles worse.

**Additional context**
### Example file
[Bridge Infill.zip](https://github.com/supermerill/SuperSlicer/files/9573302/Bridge.Infill.zip)

neophyl, 2022-09-15:
btw that layer is called the Supporting Dense Layer and its not really part of infill or bridging which is why none of the settings for those effect it I think.

jhlchu, 2022-09-15:
In gcode view it's defined as `Internal Bridge Infill`, hence I put both names in the title.

jhlchu, 2022-09-15:
The setting is called `Supporting Dense Layer` but the feature is called `Internal Bridge Infill`, as setting names don't usually match 1-to-1 with feature names.