supermerill/SuperSlicer #3217: Inaccurate seam alignment

bdwilson, 2022-09-16:
### What happened?

In order to get a completely straight seam, I have to rotate the object to have the corner towards the back of the bed and select back.  This is a shape with rounded corners (standard fillet in Fusion).  All other seam options, even painted straight line result in off seams.  I'm completely unhappy with Cura and the quality of the new version, so I've spent a ton of time getting my settings correct in SuperSlicer. The one area I'm struggling with is seams.  I have reviewed other issues (#2478 #2535 #2444 #3047 #2435) and not had any success. 

### Project file & How to reproduce

[Box with Circle.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9584048/Box.with.Circle.3mf.zip)

Nearest
[image]

Corners
[image]

Aligned
[image]

Contiguous
[image]

Rear
[image]

Rear (with object rotated - the BEST of them all)
[image]

Custom (painted seam with mouse keys to ensure straight line)
[image]
[image]

Simply change the seams in the attached project file.  You will see that the only way to get a straight seam on a corner is to rotate the object so the corner is in the back of the build plate.  This seems bonkers that none of the other options (even the corner option), can deliver a straight seam.  

### Version

2.5.59 and 2.4.58.4

### Operating system

Mac OS 12.4 (arm)

### Printer model

KP3S (using modified Hornet template in SuperSlicer)

jhlchu, 2022-09-16:
SuperSlicer_2.5.59 f8c9106
## Seam sphere
[image]
Orientation doesn't matter
## File
Don't unzip this, just rename to *.3mf.
[Box with Circle.zip](https://github.com/supermerill/SuperSlicer/files/9584418/Box.with.Circle.zip)

jhlchu, 2022-09-16:
Also better seam placement is coming once PS 2.5 changes are merge.

bdwilson, 2022-09-16:
> SuperSlicer_2.5.59 [f8c9106](https://github.com/supermerill/SuperSlicer/commit/f8c9106c0119b6373b8ace5436a756a0c0003dd8)
> 
> ## Seam sphere

Is seam sphere a new setting? I have seen it referenced but not sure how to use it.

closed 2022-09-16 by bdwilson

jhlchu, 2022-09-16:
You can also scale and rotate to get it very precise like in the file I sent back. It can be combined with seam painting for finer control. It may not always behave as you think, see #3183 for details.