supermerill/SuperSlicer #3225: Please don't implement the PrusaSlicer 2.5.0 Seams

Snuff1eupagus, 2022-09-17:
I noticed you mentioned using the seams as PrusaSlicer is using in the current version of 2.5.0 final.

Please, please, please, don't implement those seams, they are the worst seams since before version 2.3 of PS they are awful, and uncontrollable.

The seams in 2.4.1 were very much better, and as they are in SuperSlicer_2.5.59.0_win64_220811 are far better then PrusaSlicer.

I tried to create this as discussion however GitHub would not allow me to pick, I tried several times. 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

jhlchu, 2022-09-17:
Can you be more precise and give examples where the seams in PS 2.3 or 2.4.1 are better than 2.5?
Overall 2.5 seam algo leads to better alignment and prefers to put them in interior corners for less visibility.

supermerill, 2024-02-01:
quick zombie revival.
I notice the changes, and so even if I merged them, I made sure there won't be any tangible regression.
I added the "scattered" vs "random", as two kind of randomness.

Is the current version ok?

Snuff1eupagus, 2024-02-10:
I only just noticed your response today, unfortunately 2.5.59.7 win64 is acting exactly as PS 2.5 and up act now, very little seam control Sometimes in certain situations I can force the seams but rather then placing them knowingly in a few minutes, you can spend an hour trying to force them into position with no guarantee that you will achieve the desired placement.

Fresh examples,

PS 2.4 Seams
[image]

SS 2.5.59.7 Seams 

[image]

Aside from the obvious lack of control, the physical printed seams themself's look worse in PS now. I have not tested this in this latest version of SS yet.

Thanks for check in, I'm very happy to see your active here again.

supermerill, 2024-02-11:
how do you "force" them? wan I have your test projects?