supermerill/SuperSlicer #3227: Crash wehn slicing multiple objects

nirurin, 2022-09-17:
### What happened?

If I add multiple objects to the scene, variable layer height, and then slice - the program will crash to desktop. Before crashing I can briefly see a bunch of 'empty layer' error messages (there are no empty layers in the files, so this is a slicer bug). 

These empty layer messages never happened before arachne edition. I don't use arachne yet though, I'm still on classic slicer. 

It shouldn't do this. 

Newest 'arachne' version. 

### Project file & How to reproduce

[Valorant-Chamber-SniperButt_White.zip](https://github.com/supermerill/SuperSlicer/files/9591045/Valorant-Chamber-SniperButt_White.zip)

### Version

2.5.59

### Operating system

windows 11

Pavulon87, 2022-09-17:
2.5.59.0 - no problem

[image]

nirurin, 2022-09-17:
I'll try a fresh install

nirurin, 2022-09-17:
> 2.5.59.0 - no problem
> 
> [image]

Nope. Tried both 2.5.59 and the 'not arachne' 2.4.whatever. Both crash the same. 
Totally fresh downloads of superslicer, with all old files deleted. 

I can't seem to find any kind of error logs file, where is it kept? 
But yeh, definitely some jank in the code somewhere in the latest editions. I may have to roll back to 2.3 era when it worked, as I can't work with only printing one object at a time.

nirurin, 2022-09-17:
Oh I can't roll back to 2.3, it throws a bunch of pop-up errors on loading.

Reason: Failed loading configuration file "C:\Users\user\AppData\Roaming\SuperSlicer\print\0.2mm Normal @Creality 0.6 - (Custom).ini": Invalid value provided for parameter thin_perimeters: 100%

one of these for each print profile. Guess 2.4 added a new parameter. Will either have to manually edit all the config files to remove it (may try this later) or just only slice one object at a time.

nirurin, 2022-09-17:
Just did some more testing in case it was a settings issue. 

Changed printer from Ender5 to PrusaMk3s. Still crashes. 
Changed filament from my custom preset to a default. Still crashes. 
Changed print settings to a default. Still crashes. 

It will often slice the first time I hit slice, but then if I change something (especially layer heights) and slice again, it'll crash. 

Basically unusable at this point. May actually be enough to get me to go back to cura. I hate cura, but I use these printers for work.

kevinduffey, 2022-09-17:
I have the same problem.. or maybe it's a little different. SS was working great.. sliced/printed a few things from it. Then  I put some object.. forget which now.. and it was taking a bit to slice it then crashed with access violation pop up. Ever since then.. no matter what object I use.. just single objects.. it crashes. As this is not installed.. it's just a zip with an executable.. I can't uninstall and reinstall. I was hoping perhaps there was some file I delete to "reset" it.. but not sure.

nirurin, 2022-09-17:
> I have the same problem.. or maybe it's a little different. SS was working great.. sliced/printed a few things from it. Then I put some object.. forget which now.. and it was taking a bit to slice it then crashed with access violation pop up. Ever since then.. no matter what object I use.. just single objects.. it crashes. As this is not installed.. it's just a zip with an executable.. I can't uninstall and reinstall. I was hoping perhaps there was some file I delete to "reset" it.. but not sure.
> 
> Edit: So I did open up Configuration folder (Help -> Show Configuration folder). In there is SuperSlicer.ini. For the heck of it I moved those files out. Restarted SS, and low and behold.. it is slicing again.
> 
> So.. if any of the SS devs read this.. I can send you my old SuperSlicer.ini if that helps debug what might have gone wrong?

Unfortunately this did not fix the issue for me. I deleted all the config.ini files in that folder and let it remake them. Slicing still crashes if there are multiple objects on the plate. 

Have installed both cura and prusaslicer, tested, zero issues. The problem is in the superslicer code somewhere.

nirurin, 2022-09-17:
I have now completely wiped all settings and files from the userdata/superslicer directory, starting totally from scratch with no presets or filament settings. 

Still crashes. This is on a completely stock install with no custom settings or config files. App is completely broken.

nirurin, 2022-09-18:
Just as a test, I installed an older version (2.3.57) and this will slice with no issues, so the bug happened at some point since then.

nirurin, 2022-09-20:
IS there any other way to completely wipe all data for superslicer so that I can start again, as there is clearly a broken file or config file somewhere in the superslicer core which is causing it to continuously crash (it crashes on every attempt to slice multiple objects at once). I've wiped the superslicer directory, as well as the one in appdata/roaming, but this has had no effect. 

Or is there a log file somewhere which will tell me / you what the problem is? As I suspect you've just written this off as "user error" and moved on, but I'm currently stuck without a working slicer. I've installed and started calibrating Cura5 and PrusaSlicer but both of those are behind in features I commonly use.

closed 2022-09-21 by nirurin