supermerill/SuperSlicer #3236: Z-banding (wall artifacts) with Arachne enabled

pa0los, 2022-09-18:
### What happened?

When using Arachne perimeter generator there are some visible layer artifacts (similar to z-banding) on the external walls.
The same project sliced with classic generator gives smoother walls.
I have observed this on multiple different models, different filaments also.
[image]

### Project file & How to reproduce

[config.zip](https://github.com/supermerill/SuperSlicer/files/9593382/config.zip)

### Version

2.5.59.0

### Operating system

Windows 11

### Printer model

Voron Trident

liftbag, 2022-09-18:
This object is printed with arachne perimeter generator enabled, but with the following sequence: infill -> external perimeter -> perimeters. In this way the outer perimeter is extruded without any contact with the internal perimeters.
I suspect that with the default sequence, so with the outer perimeter extruded last, the variable width of the arachne perimeter generator may create more artifacts on the outer perimeter.
[image]
[image]

gaifeng8864, 2022-09-21:
> 打印此对象时启用了 arachne 周界生成器，但顺序如下：填充 -> 外部周界 -> 周界。以这种方式，外周被挤出而不与内周接触。 我怀疑使用默认序列，因此在最后挤出外周的情况下，蛛网膜周边生成器的可变宽度可能会在外周上产生更多伪影。 [image] [image]

Have you tried the following settings? I think this might be an improvement, but I'm worried that doing this might not be good for the first lap of the print overhang.
[image]

gaifeng8864, 2022-09-21:
> This object is printed with arachne perimeter generator enabled, but with the following sequence: infill -> external perimeter -> perimeters. In this way the outer perimeter is extruded without any [...]

Have you tried the following settings? I think this might be an improvement, but I'm worried that doing this might not be good for the first lap of the print overhang.
[image]

tylerinpdx, 2022-10-16:
This is not your printer I've actually seen this in the preview of my slicer as well

neophyl, 2022-10-17:
My MK3 does it with arachne in Prusa Slicer too so its not just SuSi.  I just dont think arachne is ready for the big leagues yet.  Certainly not making it the default slicing engine like PS has done.

tylerinpdx, 2022-10-17:
> My MK3 does it with arachne in Prusa Slicer too so its not just SuSi. I just dont think arachne is ready for the big leagues yet. Certainly not making it the default slicing engine like PS has done.

That's interesting what version of prusa slicer were using when I used previous slicer the newest version like yesterday it didn't have any of those artifacts in preview

neophyl, 2022-10-18:
I don't mean on the preview, I'm talking about on actual prints.  There's a thread on the Prusa forum https://forum.prusa3d.com/forum/prusaslicer/seams-issue-with-arachne/  test cube printed with arachne has z banding  and one changed to classic doesn't.

If you are seeing this in the preview then its possibly related to your xy hole compensation settings.  There are several other threads about that on here.  For example https://github.com/supermerill/SuperSlicer/issues/3292  Check your settings in your project.