supermerill/SuperSlicer #3238: Supports are great at horizontals, horrible at any other angle.

levleon13, 2022-09-19:
### What happened?

Hi. I'm a long time user of superslicer and i was able to tweak it almost perfectly to my requirements. 
The only issue left is supports that are just too far from the model at any angle between 1-85 deg. ( pic added ). 
Makes my  models look really bad when i need supports at an angle ( the only thing I'm missing from cura is three supports which are just bad ass ). 
i have tried to reduce the distance between top\bottom supports-model but no luck. only works for flat surfaces. 
[image]

### Project file & How to reproduce

just turn supports on

### Version

2..4.58.3

### Operating system

win11

### Printer model

modded cr10sv2pro

ElliottBridge, 2022-09-19:
Have you tried a finer layer height in these regions so that the stepping is less? Perhaps then the support will be close enough?

levleon13, 2022-09-19:
> Have you tried a finer layer height in these regions so that the stepping is less? Perhaps then the support will be close enough?

i did not but i dont think this will do. im printing at 0.15 layer, 0.4 nozzle, nothing crazy.

neophyl, 2022-09-19:
You should also adjust the xy separation value. The normal values are ok for up/down but when slopped it in effect pushes them further away. Same in PS.

ElliottBridge, 2022-09-19:
> You should also adjust the xy separation value. The normal values are ok for up/down but when slopped it in effect pushes them further away. Same in PS. 

Oh yeah, forgot about this myself, lol

levleon13, 2022-09-19:
> You should also adjust the xy separation value. The normal values are ok for up/down but when slopped it in effect pushes them further away. Same in PS.

That is most likely the missing key. I was set at 350%. Trying with 25% and will update! Thanks a lot.

celtare21, 2022-09-20:
I think an option to prefer the Z offset over the XY offset for interfaces would be great

inspectionsbybob, 2022-10-04:
I am having the same issue supporting the bottom of a "Bowl" shaped object.
Not only does it look awful, I think the distance (Mine is set to .2) is not being obeyed and the support structure was very very firmly attached to the bottom.