supermerill/SuperSlicer #3239: Add temperature and extrusion multiplier to modifier list

ElliottBridge, 2022-09-19:
The "issue" I am trying to solve is to vary temperature and extrusion multiplier not just in height (I don't believe extrusion multiplier can even be changed in height yet) but within a shape, like a modifier block. Why?
Well for what I want, to vary shore hardness within a part using a foaming flexible filament. So many things would be possible with the ability to change shore hardness in different areas of a print. Even engineering solutions!
As a side note, it would also offer a great degree of density control due to the foaming nature (I know infill changes can be used, but this is on top of that, it's also a side note in the sense it may be useful for others, my thought is it would be used for other foaming filaments like lightweight PLA)

I imagine the solution would be to just add the ability to modify these two things in the modifier block. It may be that the change needs to be slightly gradual within said block to avoid sharp transitions for these extruder settings, it seems possible to me, but then again I'm not a programmer.

Unfortunately for this specific need, I can't see any alternative solutions for the moment.

jhlchu, 2022-09-20:
`Temperature` and `Extrusion Multiplier` are both available for modifier shapes under the `Filament` option. However, it doesn't apply the modifier if the modifier shape only intersects with a portion of a layer. The modifier must overlap fully with a layer for it to be applied. i.e. You can't have a layer that has changing temp or EM but you can have layers with different temp or EM above or below one another.
So even though the option is available for modifier shapes, it behaves the same way as if you applied a height modifier.

ElliottBridge, 2022-09-20:
> `Temperature` and `Extrusion Multiplier` are both available for modifier shapes under the `Filament` option. However, it doesn't apply the modifier if the modifier shape only intersects with a [...]

I understand this, I was hoping it might be possible to have a setting to change the temperature within the same layer as I was describing, even if it may have to be a gradient within the modifier shape region. I know that isn't how it currently operates but could it be added as a feature?