supermerill/SuperSlicer #3241: Retraction calibration

schwirzi, 2022-09-20:
### What happened?

If i create a Retraction calibration test with a step 0,1 , 0,2 , 0,5 the steps after slicing on the right side are set to 0,1 steps every time.
only if if i use the steps 1 or 2 mm it create the correct steps.

### Project file & How to reproduce

[Retraction calibration_11m_0,1mm-1mm_210C.zip](https://github.com/supermerill/SuperSlicer/files/9607472/Retraction.calibration_11m_0.1mm-1mm_210C.zip)

### Version

2.4.58.2

### Operating system

windows 10

### Printer model

Vyper

neophyl, 2022-09-20:
Try using the . notation.  So Instead of "0,2" use "0.2"  I saw a similar bug recently in the PS github for configuration items that had the same issue.

Unfortunately for those of you in parts of the globe that use the comma notation the more readily acceptable notation is to use a full stop.  I think most code will handle . but you have to specify that , is also allowed when programming input fields rather than it being an automatic thing.

schwirzi, 2022-09-20:
i know what you mean. My settings works with , or .    . My issue is the calibrationtool itself. Today i realise that it set the steps every time to 0.1 instead of 0.1 0.2 or 0.5. 
I used on my first tests the 1mm step and see around the 2 mm retraction no stringing. the second test was 0.5mm steps and chose a high of 8 rings. so i mean it goes up to 4 mm in 0.5mm steps but it goes only up to 0.8 mm in 0.1mm steps.
so i want to tell you about a problem in the program. eventually you can fix it for other users to

RedSamura, 2022-12-13:
This problem is seen in most European languages. In English and Japanese))) - works correctly. The author has long promised to fix this error, but has not yet fixed it.