supermerill/SuperSlicer #3248: Flatpak packaging and Flathub

r0xANDt0l, 2022-09-21:
**Is your feature request related to a problem? Please describe.**
Just a simple solution for linux users who would like to auto update their Super Slicer installs easily

**Describe the solution you'd like**
Flatpak is an app packaging format that is becoming more and more widespread in the Linux Desktop, and whilst we have the AppImage, it becomes a bit annoying to go to the SS github page every time there's an update, and since most modern distros (Fedora, Mint, Crystal, to name a few) already ship flatpak support natively, it could be very easy for users in these distros to get SS

**Describe how it would work**
You could upload SS to Flathub, since it's the most used Flatpak repository, where anyone can upload their apps, and us (the users), could go to their App Stores (GNOME Software, Discover on KDE / Steam Deck, Etc...) and download it. And every time we have an update, we get a simple Update button on our app stores. 

Since there's already an AppImage of SS, You could make a flatpak of it directly, so it's less work

**Describe alternatives you've considered**
There's also the Option of using Snap Packages, but for the Linux Desktop, it isn't really liked, since it's a bit more focused at servers, but we have some desktop apps too

**Additional context**

Flathub link: https://flathub.org/home 
Flatpak Documentation: https://docs.flatpak.org/en/latest/

This would also add support for the new Steam Deck, but i doubt many people will use it in their gaming console

yordanov-alexandar, 2022-10-15:
I am using my Steam Deck as a PC as well as a gaming console and I would like to use SuperSlicer on it. Currently there is no way to use it on the Steam Deck. Both the .tar.zip and the .AppImage are giving the error 
```
[0x00007fa3d3ff0480] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
SuperSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```
And because the Steam Deck has read only file system these commands can't be run(or if they are run they will have to be run after every OS update)

redoxeon, 2022-11-01:
I'd like to use Superslicer on my steam deck as well!

cron410, 2022-12-03:
Might we adapt this code? 
https://github.com/flathub/com.prusa3d.PrusaSlicer

r0xANDt0l, 2022-12-05:
I'll try to adapt the code, and then if someone higher up wants to publish it, go ahead, although I can always publish it myself

Colcobau, 2023-03-13:
+1 for Steam Deck user who uses it as a portable computer.

GreenCappuccino, 2023-05-02:
I've repackaged the ubuntu tarball into a Flatpak:
https://github.com/GreenCappuccino/io.github.supermerill.SuperSlicer

Because I'm using the tarball, there are some idiosyncrasies, like the Desktop Integration menu still being enabled. I can't fix that right now as that's a build-time flag. We'll need to build the application like the PrusaSlicer Flatpak in the future.

Overall, the Flatpak is usable, but there's some polish that needs to be done before it can be submitted to Flathub I think.
 - Fix MIME type icon display
 - Set some other version metadata correctly
 - Tighten up certain file permissions
 - Review submission criteria for Flathub

For the time being, I'm maintaining a flatpak repo:
```sh
flatpak remote-add superslicer-flatter https://superslicer-flatter.greencappuccino.net/index.flatpakrepo
flatpak install io.github.supermerill.SuperSlicer
```

Here's how the metadata looks on GNOME Software:
[image]

I'll try to put together some better release tooling over the summer when school is out (like actually building the app). When I feel more confident about the state of the Flatpak, I'm looking to transfer ownership of the repo to @supermerill .

Vinjul1704, 2023-08-08:
fwiw, running SuperSlicer with the LC_ALL env var set to en_US.UTF-8 (`LC_ALL="en_US.UTF-8" ./superslicer`) works around the launch issue for me on my Steam Deck.

Editing the superslicer launch script and adding that part at the start of the "Exec" line works too.