supermerill/SuperSlicer #3249: Crash fault when slicing multiple items with variable layer heights

nirurin, 2022-09-21:
### What happened?

When there are multiple different objects on the print bed, and they are given variable layer heights, the whole app will crash when sliced. 

I already had an issue logged with this, but it had a bunch of unrelated information on there which wasn't going to be helpful. I've since narrowed down the problem to the most recent releases of superslicer (2.4.58.3 works fine, but the newer versions (both the arachne and non-arachne ones) both have the bug. 

I also got the windows error logs for the crashes. Each crash seems to have two logs - an access violation and an unhandled exception. 
Error Violation - https://pastebin.com/3dxZniZf
Unhandled Exception - https://pastebin.com/ipE7Zkhh

This happened regardless of printer, profile or filament. I currently have been doing tests with everything in that folder wiped, so it's using a 'default' printer and 'default' profiles (basically whatever the app things is default settings with no actual printer brand or anything set) and it still crashes when slicing. But only under the above circumstances. 

If I never printed multiple items at a time, or never uses variable layer heights, then I wouldn't have a problem. Unfortunately I do both of those things basically every print. 

I've tried to go back to cura (but am getting wierd corners and edges on finer details) and prusaslicer (works great, but no scattered infil which I need for my current projects) so for now I guess I'll be trying to go back to 2.4.58.3 for the time being and hope for the best. 

### Project file & How to reproduce

[Valorant-Chamber-Sniperpegsquare_002.zip](https://github.com/supermerill/SuperSlicer/files/9620668/Valorant-Chamber-Sniperpegsquare_002.zip)

Use any combination of any objects (as long as they are different). Make them variable layer heights. Hit slice. See the desktop appear as the app crashes. 

### Version

2.5.59.0

### Operating system

windows 11

### Printer model

doesn't matter

jhlchu, 2022-09-22:
Are you running 2.5.59 from releases or nightly builds? Because recent nightly build of 2.5.59 includes fixes for crashes from changing layer heights.

nirurin, 2022-09-22:
I'm using the main releases. I can't use nightly as they require being
built and I have no experience with doing that.

jhlchu, 2022-09-22:
It's built for you already. https://github.com/supermerill/SuperSlicer/actions/runs/3068904195
Under artifacts, click the win64 link.

nirurin, 2022-09-22:
> It's built for you already. https://github.com/supermerill/SuperSlicer/actions/runs/3068904195 Under artifacts, click the win64 link. If it works, close issue.

Interesting, didn't know that existed! And from a quick test, that seems to be working, though it's 2.4.58.4  not 2.5.59? 
It seems there's an older 'arachne fixes' nightly further down the page, I may give that a try tomorrow

jhlchu, 2022-09-22:
Versioning probably not updated. It's the 2.5-rc. The last few build have all been 2.5.59. The one linked is the latest build.
Please close issue if it's fixed.

nirurin, 2022-09-22:
The linked version doesn't have arachne, which as far as I can tell it
should have if its a 2.5?

nirurin, 2022-09-22:
> Versioning probably not updated. It's the 2.5-rc. The last few build have all been 2.5.59. The one linked is the latest build. Please close issue if it's fixed.

Just to confirm - the RC build you linked either is 2.4 (non-arachne), or it has the bug which makes the arachne toggle invisible (it certainly seems like it might have a bug, as it also then makes the arachne option invisible on even the main release 2.5). 

The 'arachne fix' nightly build from a couple days earlier than the RC works though, and it also doesn't crash with multiple objects, so it seems this problem will be resolved soon

closed 2022-09-22 by nirurin

jhlchu, 2022-09-23:
Yes, you are right. I just looked at it and the RC build is from the RC branch 2.4 and Nightly Dev is from the nightly branch 2.5. Arachne was only enabled on the 2.5 starting with the beta and the nightly builds.
The arachne fix build contained the fix for arachne seams wiping in the wrong direction.