supermerill/SuperSlicer #3252: "Only custom Start G-Code" does not work

towe96, 2022-09-22:
### Version
2.5.59.0

### Operating system type + version
Windows 11 21H2

### Behavior
SuperSlicer does not appear to honor the "Only custom Start G-Code" option in Printer Settings -> Custom G-Code

If I select above option, I expect SuperSlicer to get rid of e.g. the M104 commands for heating the printer.
This does not occur, M104 commands with the temperature selected in the Filament Settings are still added to the file.
It does not appear the matter whether I add M104 lines of my own to the start g-code, comment them out or omit them completely.

I handle start and end g-codes on my OctoPrint servers, so I don't need SuperSlicer to add superfluos commands, especially for heating since I preheat the printer before starting a print.

#### Project File (.3MF) where problem occurs
[test.zip](https://github.com/supermerill/SuperSlicer/files/9624449/test.zip)

neophyl, 2022-09-22:
Can you let us know what slicer DOESNT add temperature commands when they are set in a filament ?

Thats sort of the whole point, when printing plastic the filament settings are there to tell it what temp it needs to be.  Saying that SuSi isnt honouring the Only custom gcode is not correct, its not adding any like it used to from that section.  The ones it is adding are the perfectly normal ones done after the start gcode and during normal slicing.  You have values set outside of start gcode in the filament profile so its using them.  I don't know what else could be reasonably expected.

There are 2 things you can do.
1- If you don't want filament temp values in the gcode then update your filament profiles with zero in there.  

2- Use the post processor to remove all temp commands from the gcode.

towe96, 2022-09-22:
All other slicers I've used in the past removed "their own" heating commands when you have something like `;M104` in the start-gcode: They would recognize that as you handling the order of heating on your own and not add additional heating commands.

I'm aware of solution #1. I only noticed this bug when I imported an old Filament config which had not had that set to zero.

Still, with the wording of the option in the Printer Settings,
> Ensure that the slicer won't add heating, fan, extruder... commands before of just after your start-gcode.
led me to expect that no heating commands would be added except what I put in my manual start gcode.